PaulStoffregen/TimeAlarms #8: How to disable/cancel an event?

moarteen, 2016-03-17:
Hi,
once I issue an Alarm, for example:
Alarm.timerOnce(3600, BathReset)
how can I later disable or cancel this Alarm?
It seems like this command merely returns a true/false regarding whether the Alarm was created or not.
Looking through the code, I don't know how to get the ID, nor how to declare the variable that the ID can be contained in for future action on it.
Thanks for your help!

chrismelba, 2016-04-09:
I'd love to know the answer to this too if possible

mpena2099, 2016-04-19:
Hi folks. See this: http://forum.arduino.cc/index.php?topic=37693.0

Code:

```
#include <Time.h>
#include <TimeAlarms.h>

#define TMR_LOGGER  60
AlarmId loggit;           // create loggit as a global variable

void setup(){
 Serial.begin(9600);
 loggit = Alarm.timerRepeat(TMR_LOGGER, logTime);     // create log scheduler alarm and store the id in loggit
}

void logTime(){
 Serial.println("Alarm");    
}

void  loop(){  
 
}

void fullLog() {                                     // Stop logging when full
addr = 0;                                           // reset address pointer
Alarm.disable(loggit);                          // shut off logTime alarm
}
```

So... we need to update the documentation...

commit e82df200 (2016-07-06): Update example to demonstrate Alarm.free(id)
fixes #8
closed 2016-07-06 by PaulStoffregen