PaulStoffregen/TimeAlarms #9: compile errors when TimeAlarms.h is included

mckennatim, 2016-03-19:
Looks like a great library but I get these errors as soon as I include it for my esp8266 project:Any work arounds? ` arduino 1.67, wemos mini esp8266, Time v1.5.0, TimeAlarms v1.4.0`

In file included from sketch\ProgTimer.cpp:3:0:

C:\Users\tim\Documents\electronics\arduino\libraries\TimeAlarms/TimeAlarms.h:71:31: error: 'timeDayOfWeek_t' does not name a type

   AlarmID_t alarmRepeat(const timeDayOfWeek_t DOW, const int H,  const int M,  const int S, OnTick_t onTickHandler); // as above, with day of week 

                               ^

C:\Users\tim\Documents\electronics\arduino\libraries\TimeAlarms/TimeAlarms.h:71:47: error: ISO C++ forbids declaration of 'DOW' with no type [-fpermissive]

   AlarmID_t alarmRepeat(const timeDayOfWeek_t DOW, const int H,  const int M,  const int S, OnTick_t onTickHandler); // as above, with day of week 

                                               ^

C:\Users\tim\Documents\electronics\arduino\libraries\TimeAlarms/TimeAlarms.h:75:29: error: 'timeDayOfWeek_t' does not name a type

   AlarmID_t alarmOnce(const timeDayOfWeek_t DOW, const int H,  const int M,  const int S, OnTick_t onTickHandler); // as above, with day of week 

                             ^

C:\Users\tim\Documents\electronics\arduino\libraries\TimeAlarms/TimeAlarms.h:75:45: error: ISO C++ forbids declaration of 'DOW' with no type [-fpermissive]

   AlarmID_t alarmOnce(const timeDayOfWeek_t DOW, const int H,  const int M,  const int S, OnTick_t onTickHandler); // as above, with day of week 

                                             ^

exit status 1

PaulStoffregen, 2016-03-19:
Does this same error happen with regular Arduino?

If not, perhaps it's a problem with ESP8266?  I have never used ESP8266 and generally do not support it when the code works properly with Arduino.

The-Alchemist, 2016-06-30:
I just wanted to confirm the original problem, just on a different board: Adafruit Feather M0 (Arduino Zero variant).

PaulStoffregen, 2016-07-06:
Fixed by Time.h to TimeLib.h change

closed 2016-07-06 by PaulStoffregen