PaulStoffregen/TimeAlarms #18: Parameters

mal2tin, 2016-09-12:
I reallly need a form to use parameters.. 
Alarm.timerOnce (10,0,0, ledOn (2));
Void ledOn (int ledPin) {
digitalWtite (ledPin, High) 
} 

Can you help me?

Modifikator, 2017-04-02:
I too have experienced this problem. Has anyone resolved this issue?

justinatomatic, 2017-08-10:
I'd love to see this as well. 
Might be easier to do timerOnce(const int H,  const int M,  const int S, OnTick_t onTickHandler, int data);

Pass the data/parameter separately store it in the AlarmClass and pass it back to the callback function.

I was hoping somebody might have already extended TimeAlarms like this.

riskiadi, 2019-03-18:
@Modifikator @mal2tin @justinatomatic 
if you guys confused, try this another way, WITHOUT "AlarmTimer" Library! but need Epoch value
`int phase = 1;
void setup(){

}
void loop(){
          int8_t hours  = epoch.toInt() / 3600 % 24;
          int8_t minute = epoch.toInt() / 60 % 60;

          if(hours   == 17 & minute == 30){
            if(phase == 1){
              dataTraining(temp, humidity, light, moisture, epoch);
              phase++;
            }
          }else if(hours == 17 & minute == 31){
           if(phase == 2){
              dataTraining(temp, humidity, light, moisture, epoch);
              phase++;
            }
           }else if(hours == 17 & minute == 33){
           if(phase == 3){
              dataTraining(temp, humidity, light, moisture, epoch);
              phase=1;
            }
           }
}`

spaelectronics, 2023-03-28:
I also need to be able to pass variables to the function. Is this possible?