PaulStoffregen/TimeAlarms #21: Wemos D1 mini: Alarm.delay() problems is more than 1000 ms

lsteffenel, 2016-11-11:
I'm using a Wemos D1 mini and the board resets if I set Alarm.delay() with more than 1000 ms. The only way to make longer pauses is to put a loop with {Alarm.delay(1000);delay(1);} as even two consecutive Alarm.delay(1000) reset the board. 

I also have a second question, is TimeAlarms compatible with deepsleep? The source code for Alarm.delay() seems a busy-wait loop, which is not very energy-saver...

luckyhacky, 2017-07-16:
see issue https://github.com/PaulStoffregen/TimeAlarms/issues/23

I had the same issue.
Reason for soft watchdog is, that the delay function of TimeAlarm lib is activly waiting in loop and calling serviceAlarms() function. The reset of watchdog is not triggered. This can be avoided either by splitting code into two portions:

```
Alarm.delay(5000); 
```
```
Alarm.delay(500); 
delay(4500);
```

or by using my proposed fix, which does it automatically.
see https://github.com/PaulStoffregen/TimeAlarms/pull/29

gschintgen, 2018-02-14:
Hi, I don't think using the regular delay() function is the right solution: alarms are only serviced during the 500 first milliseconds. It's better to call delay(0) or yield() (which is the same) in every loop iteration.
I submitted a pull request implementing this alternative approach: see #34 .

luckyhacky, 2018-02-25:
seems are more professional fix than my quickfix.
should be pulled to mainline :)

PaulStoffregen, 2018-03-16:
fixed by #34

closed 2018-03-16 by PaulStoffregen