PaulStoffregen/TimeAlarms #23: NTP Problem on ESP8266-E12

bobybobybob, 2017-02-04:
Hello,

If I use the example script, everything goes great. Also on my Arduino Uno with RTC everything is super.

For an ESP8266 with NTP without RTC the alarm works fine every 15 seconds. Does not work does alarm at certain times. For alarm time I use UTC and UTC time is displayed with digitalClockDisplay (); In serial print, correct UTC time is displayed. No alarm at alarm times.

I use that NTP Script for EPS8266-E12 to get Time from Internet
http://www.arduinoclub.de/2016/05/07/arduino-ide-esp8266-ntp-server-timezone/

Do you have a idea?

PaulStoffregen, 2017-02-06:
Nope, I do not know.  I do not use ESP chips.  I'm depending on the ESP community to resolve any ESP-specific problems and submit high quality pull requests (low risk for non-ESP).  Sorry, I just can't spend time working for ESP.

ozarchie, 2017-03-06:
Hi Paul,
Not sure if this helps:

Having a similar problem on NODEMCUs (ESP8266).
The use of Alarm.delay(val) cause a fault if 'val' is set  too high. 

At the end of my loop I inserted the following code:

```
int AlarmDelay = 1500;

.
.

  Alarm.delay(AlarmDelay+=1);
  DEBUG_print("Loop - ");
  DEBUG_print(AlarmDelay+=1);
  DEBUG_print(" : ");
  showDateTime();
```

Result: It fails when AlarmDelay exceeds 1622

```

...

Loop - 1613 : 13:37:30 6/3/2017
Loop - 1614 : 13:37:31 6/3/2017
Loop - 1615 : 13:37:33 6/3/2017
Loop - 1616 : 13:37:34 6/3/2017
13:37:35 6/3/2017
Loop - 1617 : 13:37:36 6/3/2017
Loop - 1618 : 13:37:38 6/3/2017
Loop - 1619 : 13:37:39 6/3/2017
Loop - 1620 : 13:37:41 6/3/2017
Loop - 1621 : 13:37:43 6/3/2017
Loop - 1622 : 13:37:44 6/3/2017

Soft WDT reset

ctx: cont 
sp: 3fff1cd0 end: 3fff1f20 offset: 01b0

>>>stack>>>
3fff1e80:  3ffe98e7 000007e1 3fff0e3c 4020dc17  
3fff1e90:  3ffe9e32 3ffe9e32 3fff0bcc 4020a89c  
3fff1ea0:  3fff0acc 3fff0acc 3fff0bcc 4020ac50  
3fff1eb0:  3ffe9668 00000000 000003e8 00000001  
3fff1ec0:  3fff0acc 00037ac1 00000657 4020accc  
3fff1ed0:  3fffdad0 3ffe8540 3fff0e3c 4020518d  
3fff1ee0:  00000005 3ffefeba feefeffe feefeffe  
3fff1ef0:  00000000 00000000 00000001 3fff0ef8  
3fff1f00:  3fffdad0 00000000 3fff0ef0 4020f100  
3fff1f10:  feefeffe feefeffe 3fff0f00 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

luckyhacky, 2017-07-16:
I had the same issue.
Reason for soft watchdog is, that the delay function of TimeAlarm lib is activly waiting in loop and calling serviceAlarms() function. The reset of watchdog is not triggered. This can be avoided either by splitting code into two portions:

```
Alarm.delay(5000); 
```
```
Alarm.delay(500); 
delay(4500);
```

or by using my proposed fix, which does it automatically.
see https://github.com/PaulStoffregen/TimeAlarms/pull/29

pierangelof, 2017-07-27:
Hi bobybobybob,
have you tried a solution for the alarms with an ESP8266 board? I have found the following modification of the library of Paul
https://github.com/PaulStoffregen/TimeAlarms/pull/16/files
and it works.

bobybobybob, 2017-09-19:
No i have not tried a solution with time alarms. I have programmed a primitive replacement. It works as desired, but TimeAlarms is more elegant and easier to use. Crying that it is not compatible for ESP 8266.

bradyjoh, 2017-11-12:
I too have this issue.

curlyz, 2018-07-06:
Ok , you can't do it directly like that , here are my dirty method that I use
The simpliest way
1. Create an non-overflown time count. Hint . Using a period timer to add ticks or something, on Python I don't need to care about number overflowing but you should
2. Sync NTP only once , save that timestamp along with the runtime (from the non-ovf timer)
3. When you need to add an alarm , calculate the future runtime by using the value from step 2 , you will know how long does it take till your next alarm in milliseconds.
4 . Sync the delay on the ESP is limited , you can't call delay(timeleft). You must use a timer to periodic calling a handler function to that chop the time into smaller section, let say your alarm is 50 minutes later.  The timer will be called at 15,30,45,50 . The last call (50) will set a flag for the main loop to run , the others (15,30,45) are there to counter the timer max period .

romuye123, 2018-11-28:
Hello,
I am facing the same problem too on ESP8266 12F. this happens when any timer or schedule is triggered.
I have kept **Alarm.delay(0);** at the end of my loop function.
Also added  **yield();** as per [bd95110](https://github.com/PaulStoffregen/TimeAlarms/commit/bd9511040460c41be72d899899f48a7ead436708)

Get following error:

```
Soft WDT reset

ctx: sys 
sp: 
Exception (
Exception (
Exception (
Exception (
Exception (
Exception (
Exception (
Exception (
Exception (
Exception (
Exception (
Exception (
Exception (
 ets Jan  8 2013,rst cause:4, boot mode:(3,1)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
```

jgachite, 2021-06-29:
The problem is Alarm.delay
Use delay(xxx);