PaulStoffregen/TimeAlarms #24: Not working with ESP8266+RTC3231

bkrajendra, 2017-02-20:
I'm using this library with ESP8266 and RTC Module DS3231.
```
#include <ESP8266WiFi.h>
#include <Wire.h>
#include <TimeLib.h>
#include <TimeAlarms.h>
#include <RTClib.h>
AlarmId id;
RTC_DS3231 rtc;

void setup()
{
    DateTime now = rtc.now();
    setTime(now.hour(),now.minute(),now.second(),now.day(),now.month(),now.year()); // set time
     //time_t t = now.unixtime(); // tried both method of setting time.
     //setTime(t);

      if (! rtc.begin()) {
        Serial.println("Couldn't find RTC");
        while (1);
      }
    id = Alarm.alarmRepeat(16, 02, 0, MorningAlarm);
    pinMode(2, OUTPUT);
   digitalWrite(2, LOW);
}
void MorningAlarm() {
  Serial.printf("Alarm Running..!");
  digitalWrite(2, HIGH);
}

void loop() {

}
```
What could be the issue.?

PaulStoffregen, 2017-02-20:
I do not use ESP, so I can not assist.  Maybe you should ask for help on the ESP community forums?  This github issue tracker is for reporting confirmed bugs.  Please do not use it for technical support on your project.  Many things do not work properly on ESP chips.  Only post here if you are sure, or if an expert has confirmed the problem is due to a bug in this library.

closed 2017-02-20 by PaulStoffregen

bkrajendra, 2017-02-20:
Thank you @PaulStoffregen for reply.. 
I do understand your perspective, but don't you think if anyone answers this question here, it will be great for anyone searching for this answer. Do you know how difficult it is to find this answer on ESP related forum as its very big and scattered community.

I was at least expecting to know if I'm doing anything wrong in general. I'll definitely post answer here if I  resolve it. 
Sorry for trouble. Thank You!