PaulStoffregen/TimeAlarms #28: Alarms do not fire when time is changed

kdhooper, 2017-06-29:
When changing the time in the Time library, TimeAlarms misses alarms.

The expected behavior is:
```C
update() {
  if(last_update_time < alarm_time < present_update_time) {
    TriggerAlarmCallback();
  }
}
```
..in other words, every instance of the time passing through a recurring alarm should result in the alarm firing.

You can demonstrate this bug by
* Set the time for 7:00AM,
* Set a recurring alarm for 7:02AM
* Watch it trigger
* Set the time again to 7:00AM
* The alarm will not trigger the second time

A more practical example of this bug is if the time is set to 7:00PM and you run the clock back to 7:00AM, the alarm will not trigger. This is highly likely to cause alarms (possibly for the whole day) to be missed, and is something a user of an alarm clock might do.

Workaround:
Update the alarm each instance of updating the time:
```C
void saveTimeCurrentTime(uint8_t setHour, uint8_t setMinute) {
  setTime(setHour, setMinute, 0, day(), month(), year());
  Alarm.write(gAlarmID1, AlarmHMS(gTime1.hours, gTime1.minutes, 0));
}
```
Where `gTime1` struct holds the hour and minute of the alarm as a global variable.

luckyhacky, 2017-07-16:
I have the same issue. Alarms are omited if NTPClient updated time.

luckyhacky, 2017-07-16:
I found my problem. Just a short sketch for everybody using a similar code:
```
int ntp_time = 0;
NTPClient ...

void setup()
{
setSyncProvider(ntp_set_time);
...
}

void loop()
{
...
}

long ntp_set_time()
{
  while (ntp_time < 1000000)
  {
    ntpClient.update();
    ntp_time = ntpClient.getEpochTime();
    delay(1000);
  }

  Serial.print(ntpClient.getFormattedTime());
  Serial.print(" [TIME ] Time to set : " );
  Serial.println(ntp_time);

  return ntp_time;
}
```

Issue arises, due to ntp_time being global. So time is not updated.
A local variable ntp_time in function ntp_set_time would block, if connection to NTP servers fails. 

My current solution:
```
long ntp_set_time()
{
  ntpClient.update();
  ntp_time = ntpClient.getEpochTime();
  
  while (ntp_time < 1000000)
  {
    ntpClient.update();
    ntp_time = ntpClient.getEpochTime();
    delay(1000);
  }

  Serial.print(ntpClient.getFormattedTime());
  Serial.print(" [TIME ] Time to set : " );
  Serial.println(ntp_time);

  return ntp_time;
}
```

This is just for information and not related to bug mentioned above.

luckyhacky, 2017-07-16:
A proposal for retriggering as you suggested may be https://github.com/luckyhacky/TimeAlarms/commit/9b198a92d2793eeab041757b3b086819f9184833
you can test and give feedback.

Janovlk, 2025-12-25:
Sometimes I found that timers weren't triggered after updating the time anymore. The solution was to call a function (Alarm.delay) right after the time was updated and not only keep it in loop() only. For convenience I created a new function Alarm.loop which is the Alarm.delay function minus the delay functionality.