PaulStoffregen/TimeAlarms #30: getNextTrigger does not work on Wemos D1

maoterodapena, 2017-07-17:
Hi. getNextTrigger always returns -1 on Wemos D1 (I have not tested it on another micro). I've noticed that the loop on the allocated alarms starts with nextTrigger = ffffffff, which is -1, so it never updates to a minor value in the inner comparation.

If I put 7ffffff as the maximun value the function returns the right result, but I'm not sure there is another hidden conflict.

Scannow, 2018-01-02:
Same issue here.

"**time_t**" has the wrong type declared in the "**types.h**" file, inside the esp8266 sdk.
Open the file
_%LOCALAPPDATA%\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\machine\types.h_
and change the line #14 from
`#define	_TIME_T_	long`
to
`#define	_TIME_T_	long unsigned int`

kaloskagatos, 2018-02-10:
Hi. The norm does not define the sign of time_t. 
https://stackoverflow.com/questions/471248/what-is-ultimately-a-time-t-typedef-to

So setting nextTrigger to ffffffff overflows and getNextTrigger never meets the condition to find an alarm. One solution would be to assign the max value of a time_t to the nextTrigger variable in TimeAlarmsClass::getNextTrigger(). But <limits> should not be available on arduino.

```
#include <limits>
...
time_t TimeAlarmsClass::getNextTrigger()
{
   // time_t nextTrigger = (time_t)0xffffffff;  // the max time value
   time_t nextTrigger= std::numeric_limits<time_t>::max();
```