PaulStoffregen/TimeAlarms #36: alarmOnce(H,M,S,Handler) and others won't fire at midnight

gritnix, 2018-03-31:
### Description

alarmOnce and presumably alarmRepeat where HMS is given does not fire at midnight.

### Steps To Reproduce Problem

set alarmOnce(0,0,0,myHandler);

This will not fire.  It's easy to see why and if I get a chance I'll update .h and do a pull request.  Certain functions, most notably

AlarmID_t alarmRepeat(const timeDayOfWeek_t DOW, const int H, const int M, const int S, OnTick_t onTickHandler) 

and 

AlarmID_t alarmOnce(const timeDayOfWeek_t DOW, const int H, const int M, const int S, OnTick_t onTickHandler)

and

AlarmID_t alarmOnce(const int H, const int M, const int S, OnTick_t onTickHandler)

all call AlarmHMS() and then call their time_t variant with the result.  The time_t variant does this check

if (value <= 0)  return invalid alarm.

Passing in midnight, 0,0,0, gives a value of 0 which is perfectly ok.  This needs to be changed to < 0, not <=.

Thanks.

Gerry33, 2018-06-05:
@gritnix  : Same problem here.
But your suggested solution wonn't  fix the problem. With your suggested patch the timer is accepted, but never executed.  Problem is somewhere deeper in the logic.

This is never executed:
`  Alarm.timerOnce (0,  OnceOnly);   `

But I see no real reason why this shouldn't. It simply means, 'immediately'. 

Thanks anyhow 
Gerry

countrysideboy, 2020-02-21:
Try a temporary way to fix it:
Set the alarm from 00:00:00 to 00:00:01

countrysideboy, 2020-02-21:
The AlarmHMS() macro does not deal with the 00:00:00。If the hour/min/sec is all zero, it should return the seconds of (00:00:00-now_time),instead of zero.

And the timealarm.ccp deal with the overnight condition.

gonzzzales, 2026-10-19:
```
  AlarmID_t alarmOnce(const timeDayOfWeek_t DOW, const int H, const int M, const int S, OnTick_t onTickHandler) {
    time_t value = (DOW-1) * SECS_PER_DAY + AlarmHMS(H,M,S);
    //if (value <= 0) return dtINVALID_ALARM_ID;
	if (value < 0) return dtINVALID_ALARM_ID;
    return create(value, onTickHandler, true, dtWeeklyAlarm);
  }
```
value = 0 when DOW = 1 (Sunday) and H,M,S are 0. I does this
```
if (value < 0) return dtINVALID_ALARM_ID;
```