PaulStoffregen/TimeAlarms #42: delay might fail after ~50days runtime

VeryPrivat, 2018-10-07:
Analyzing your code shows following problem in line 195 of TimeAlarms.cpp: 
* millis() will overrun after 49,7 days since it is an unsigned long (4.294.967.295 / 86.400.000 [ms/day] = 49,710 days)
* If start + ms is bigger than 4.294.967.295 calculation of millis() - start will never reach the value of ms =>  While loop will never end

tonilopezmr, 2019-11-29:
@VeryPrivat @b4dnewz @crisap94 is this already fixed?

b4dnewz, 2019-11-30:
@tonilopezmr I don't know, I don't even know if is a real issue
> When using subtraction on unsigned integers. the subtraction will automatically handle it.
 millis() - startTime >= interval   Lets try it using byte to make the math easier
 10  - 245 >=  20   Millis() has just rolled over startTime was before rollover
 if you start at 10 count back to 0 then 255 on down you will see that the answer is exactly 20.
subtraction will always give the right answer when using unsigned integers hence no problems with rollover.

source; https://forum.arduino.cc/index.php?topic=543895.0