PaulStoffregen/TimeAlarms #45: Alarm.alarmRepeat() won't trigger after setTime

udivankin, 2018-11-24:
I thinks that must be either prominently mentioned in README or got fixed.

**This code doesn't work as expected:**
`
#include <Time.h>
#include <TimeAlarms.h>

void setup() {
  Serial.begin(9600);
  Alarm.alarmRepeat(8,29,30, MorningAlarm);
  setTime(8,29,25,1,1,11);
}

void test(){
  Serial.println("test");    
}

void loop(){  
  Alarm.delay(1000);
}
`

**This code also doesn't work:**
`
#include <Time.h>
#include <TimeAlarms.h>

AlarmID_t  alarm1;

void setup() {
  Serial.begin(9600);
  alarm1 = Alarm.alarmRepeat(8,29,30, MorningAlarm);
  setTime(8,29,25,1,1,11);
  Alarm.enable(alarm1);
}

void test(){
  Serial.println("test");    
}

void loop(){  
  Alarm.delay(1000);
}
`

**But this one works!:**
`
#include <Time.h>
#include <TimeAlarms.h>

AlarmID_t  alarm1;

void setup() {
  Serial.begin(9600);
  setTime(8,29,25,1,1,11);
  alarm1 = Alarm.alarmRepeat(8,29,30, MorningAlarm);
}

void test(){
  Serial.println("test");    
}

void loop(){  
  Alarm.delay(1000);
}
`

rewida17, 2019-07-08:
How I can add second alarm ? Just "AlarmID_t  alarm1;"  define EveningAlarm for eq?

tonilopezmr, 2019-11-29:
@udivankin You can't set the time after setting the alarm because TimeAlamrs check if the actual time is valid (which is not, because you didn't set the time). 

If you print `alarm1` you will see `255` which is the `invalidAlarm` id

sjernigan, 2021-03-26:
This bit me too.  

BEWARE: time is not epoch time but seconds since 1990.

I was using an NTP sync provider but removed the offset to change the time from epoch time to "seconds since 1990" when adding an automatic timezone calculation.  Anyhow the year was set in the future and none of my alarms would set because they were in the past.  I'm not saying this is a bug cause it would have also taken me a long time to figure out why my alarms were firing.  The only change I might suggest would be to return a couple of different error codes (time not set, alarm in past, invalid time).