PaulStoffregen/TimeAlarms #46: SOFT WDT RESET

romuye123, 2018-11-29:
### Description

TimeAlarms gives following exception when using along with EEPROM or SPIFFS.

```
Soft WDT reset

ctx: sys 
sp: 
Exception (
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

### Steps To Reproduce Problem

1. Create alarms and save them in eeprom or SPIFFS.
2. Say after a minute or two, ESP8266 gives a SOFT WDT RESET as shown above.

### Hardware & Software

Board : GENERIC ESP8266
Shields / modules used: NODEMCU
Arduino IDE version : 1.6.5
Version info & package name : 2.4.2

##

udivankin, 2018-11-29:
Could you share minimal reproducible example here?
I personally have no problems using TimeAlarms with EEPROM

romuye123, 2018-11-30:
Yes you are right. Seems to be issue with placing of Alarm.delay() in your loop function.  

As I read somewhere in issues list that it needs to be put at the end of the loop function so I did it accordingly and it was giving me soft reset.

closed 2018-11-30 by romuye123