PaulStoffregen/TimeAlarms #51: Features request

countrysideboy, 2020-02-17:
I use the fake code below to trigger a hourly bell.

1.Settime(...)

2.(add 48 alarms)
Alarm.repeate(00:00, bell)
Alarm.repeate(00:30,bell)
....
Alarm.repeate(23:00, bell)

3.
Void bell()
{
...
}

I think the 48 alarms may use the small ram of esp8266, could you please add a graceful function that can alarm hourly all the day?
Thank you.

countrysideboy, 2020-02-18:
Finally I use that fake code with just one alarm.
I also put the service () method into public so that I can call Alarm.service() in loop instead of Alarm.delay().
(I use mobile to edit the text, maybe I write the wrong function name, I am sorry for that and I hope u could understand what I mean.)

1.settime

2.
Void set_next_alarm()
{
int next_hour= hour();
If (next_hour==23) next_hour=0;
else next_hour +=1;
Alarm.onec(next,0,0,alarm_function);
}

3.
Void alarm_function()
{
  set_next_alarm();
......Do any thing you want.
}

Jediweirdo, 2024-04-27:
If you want to clean this up in the future, use `Alarm.timerRepeat()` lol. Try something like this and tell me how it goes:

```
Alarm.alarmOnce(hour() + 1, 0, 0, initiateHourlyTimers); //<- Garentees that the alarm goes off at the top of the next hour

void initiateHourlyTimers() {
Alarm.timerRepeat(3600, bell); //<-- 3600 seconds = 1 hour. Will go off every hour without the need to manually recreate it everytime
}

void bell() {
// idk bell code here
}
```