PaulStoffregen/TimeAlarms #59: can't make Alarm.alarmRepeat work

donbrew1, 2021-10-09:
For unknown reasons I can't create an account at forum.arbuino.cc

I could not get the alarmRepeat function to work in my sketch, so I loaded the example (10/9/2021); it still does not work.  The only thing I changed was my network and password and time zone..
Tried both ESP8266TimeAlarms.h and TimeAlarms.h.  (example from each).
Clock time is correct.

Board esp8266.  D1 mini and NodeMCU 
4M(1M spiffs) but tried others.
Timers work. Alarms don't

Arduino IDE version: 1.8.13

Grandpa-G, 2022-10-26:
I have to agree. I am using an ESP32 and alarmRepeat and triggerOnce don't ever fire.

Grandpa-G, 2022-10-27:
I finally got the alarms to work. 2 things are critical,

Alarm.delay(1000);
AND
setTime(T);

The lack of setTime was preventing my alarms from working.